NachtRaveVL/Lepton-FLiR-Arduino #1: A few problems

jake-g, 2017-10-28:
I have a teensy 3.6 and a flir v1.4 module and am trying to send the frame pixels over serial and plot in python.

I've run into a few issues, and I have time to help resolve them , but I will need some assistance:

1. `setSysTelemetryEnabled` is not defined in the library, so the     `flirController.setSysTelemetryEnabled(ENABLED); ` does not work. I'm not really sure what this even does. commenting out the above telemetry line lets me run some of the examples

2. I tweaked the advanced example to serial print all of the pixels in each frame. When I plot these as an image, it doesn't look correct. Looks like random noise. I can provide an example if need be.

3. I have issues with the write to SD card example. It seems to hang and never write anything. I wanted to test this to see if issue 2 above is my fault, or if the written sd card images also look this way.

thanks

jake-g, 2017-11-01:
fixed some of this

closed 2017-11-01 by jake-g

derekwacks, 2019-07-10:
In the library file LeptonFLiR.h, it's defined as
 `void sys_setTelemetryEnabled(bool enabled);`
so the line in the adv. example should be changed to
`flirController.sys_setTelemetryEnabled(ENABLED); `

NachtRaveVL, 2019-07-10:
Oh wow, thanks for the heads up. If you’d like to make a pull request for this I’d really love the help, otherwise I will make one later today when I get back home from work.

Thanks for the interest!

I do ask tho, have you been able to get it to work on a teensy 3.6? I have one as well, I should probably try it out and see if the software copy is fast enough with it.

derekwacks, 2019-07-11:
There should be an open pull request from @samyk that resolves it. I hadn't tried it on a teensy, though I might have one laying around to test it. I connected it to an arduino Mega, but realized there isn't enough capacity to capture a full frame, so I'm switching to a rPi 
Thank you so much for the library!