NachtRaveVL/Lepton-FLiR-Arduino #4: flirController.readNextFrame() always returns false?

ojohnson2, 2019-12-11:
I'm using an arduino Uno and am attempting to utilize this code however I cannot get flirController.readNextFrame() to return true so I can actually utilize the camera. I recognize that I am inexperienced with arduino but I believe I have everything wired correctly as it is plugged directly into the board with the pins matching the corresponding functions.

closed 2019-12-12 by ojohnson2

NachtRaveVL, 2019-12-12:
Unfortunately a UNO will not be powerful enough to correctly grab the frame in time before the sensor invalidates the feed.

I couldn’t even get this to work fast enough with a Duo, and would suspect you’d need at least a Teensy 3.4 or better to even get this library working in any way. *le sigh*

ojohnson2, 2019-12-12:
I appreciate the information, as unfortunate as it may be. But hey, saved me some time and a headache.

GoncaloRodrigues241, 2021-06-09:
Hello!

I have the same problem, flirController.readNextframe() is always returning false and I am using an Adafruit ESP32 Feather.

How can I solve it?

Tkank you