NachtRaveVL/Lepton-FLiR-Arduino #7: Read Next Frame not working on ESP32

techboycr, 2020-08-20:
Awesome initiative thanks a lot!

I am trying to run it in ESP32 (Sparkfun ESP32 Thing Plus) but is not getting the frames This is the debug output with one get frame.

`Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
LeptonFLiR::init spiCSPin: 21, storageMode: 3, tempMode: 0
  LeptonFLiR::init imageData: 1447B, spiFrameData: 367B, total: 1814B
  LeptonFLiR::init SPIPortSpeed: 15.00MHz (SPI_CLOCK_DIV16)

 ~~~ LeptonFLiR Module Info ~~~

Chip Select Pin:
D21 (active-low)

SPI Port Speed:
15.00MHz (SPI_CLOCK_DIV16)

Image Storage Mode:
3: LeptonFLiR_ImageStorageMode_40x30_8bpp

Temperature Mode:
0: LeptonFLiR_TemperatureMode_Celsius

Memory Footprint:
Image Data: 1447B, SPI Frame Data: 367B, Telemetry Data: 0B, Total: 1814B

Power Register:
    LeptonFLiR::readRegister regAddress: 0x0
      LeptonFLiR::readRegister retVal: 0x0
0x0

Status Register:
    LeptonFLiR::readRegister regAddress: 0x2
      LeptonFLiR::readRegister retVal: 0x6
0x6

AGC Enabled:
LeptonFLiR::agc_getAGCEnabled
  LeptonFLiR::receiveCommand cmdCode: 0x100
    LeptonFLiR::waitCommandBegin
      LeptonFLiR::readRegister regAddress: 0x2
      LeptonFLiR::readRegister retVal: 0x6
    LeptonFLiR::writeRegister regAddress: 0x4, value: 0x100
    LeptonFLiR::waitCommandFinish
      LeptonFLiR::readRegister regAddress: 0x2
      LeptonFLiR::readRegister retVal: 0x6
      LeptonFLiR::readDataRegister readWords[2]: 0x0-0x0
disabled

AGC Policy:
LeptonFLiR::agc_getAGCPolicy
  LeptonFLiR::receiveCommand cmdCode: 0x104
    LeptonFLiR::waitCommandBegin
      LeptonFLiR::readRegister regAddress: 0x2
      LeptonFLiR::readRegister retVal: 0x6
    LeptonFLiR::writeRegister regAddress: 0x4, value: 0x104
    LeptonFLiR::waitCommandFinish
      LeptonFLiR::readRegister regAddress: 0x2
      LeptonFLiR::readRegister retVal: 0x6
      LeptonFLiR::readDataRegister readWords[2]: 0x1-0x0
[... 182 lines cut ...]
      LeptonFLiR::readRegister regAddress: 0x2
      LeptonFLiR::readRegister retVal: 0x6
    LeptonFLiR::writeRegister regAddress: 0x4, value: 0x218
    LeptonFLiR::waitCommandFinish
      LeptonFLiR::readRegister regAddress: 0x2
      LeptonFLiR::readRegister retVal: 0x6
      LeptonFLiR::readDataRegister readWords[2]: 0x1-0x0
  LeptonFLiR::receiveCommand cmdCode: 0x21C
    LeptonFLiR::waitCommandBegin
      LeptonFLiR::readRegister regAddress: 0x2
      LeptonFLiR::readRegister retVal: 0x6
    LeptonFLiR::writeRegister regAddress: 0x4, value: 0x21C
    LeptonFLiR::waitCommandFinish
      LeptonFLiR::readRegister regAddress: 0x2
      LeptonFLiR::readRegister retVal: 0x6
      LeptonFLiR::readDataRegister readWords[2]: 0x1-0x0
    LeptonFLiR::readRegister regAddress: 0x2
      LeptonFLiR::readRegister retVal: 0x6
  LeptonFLiR::readNextFrame AGC-8bit: disabled, Telemetry: enabled, Location: footer
  LeptonFLiR::readNextFrame Maximum resync retries reached. Aborting.
`

I dont know if there anything that I am doing wrong. 

Thanks for your help.

NachtRaveVL, 2020-08-20:
Heyhey,

Thank you for testing this on the ESP32 and filing out an issue. This is actually a known issue as of currently, due to the slow SPI interface that the Arduino libraries utilize. A more thorough explanation of what’s happening is on the other issue: https://github.com/NachtRaveVL/Lepton-FLiR-Arduino/issues/2

Essentially, you’re experiencing a de-sync and that’s not surprising given that this has long since been the major bottleneck of the library.

I’ve been working on a new version of the library and a completely different fast SPI transfer mechanism in my spare time as of more recently. I don’t have a time table for any of it, as I have only time in the evening after the day job to work on such, but I would recommend instead to go the route of RasPi to utilize the FLiR as of currently.

On a different note: did you experience any compilation difficulties with compiling the library on ESP32? I admit I haven’t looked too far into the Espressif toolkit there (although I did install it for sake of future experimentation), as I’ve been busy with a lot of other projects lately.

Thanks again!
NR

techboycr, 2020-08-21:
Hello Hello!

Well this piece of software you did is just amazing, thanks for that!  I will give it a different try tonigh and I will keep you inform on my findings.

And to answer your not I didn’t experience no problems at compiling time, So that led me to think that maybe we will be able to make it work soon.

Thanks for your help!

closed 2020-08-21 by techboycr