NachtRaveVL/Lepton-FLiR-Arduino #8: How to wiring on ESP32?

tomeroto, 2020-09-14:
Hi,

Thanks for you libraries.

Btw, I was testing my ESP32 with this setup pin:

```
ESP32    | FLIR LEPTON
3.3V     | VIN
GND      | GND
18       | CLK
19       | MISO
23       | MOSI
27       | CS
```

But it is nothing happen on serial output. Do you have recommended wiring / pin setup for ESP32?

NachtRaveVL, 2020-09-21:
I'll have to wire it up on the ESP32 and give it a go - more than likely, you'll need to make sure Wire1 is properly defined, and properly set up. I was looking at supporting this in the next version, but I admit I haven't gotten around to fully testing it yet. It's on my todo list, albeit I've been a bit preoccupied with general life stuff.

imaurus, 2021-01-09:
I'm working on this. I managed to make it work with my ESP32 dev kit.
Here the pinout:

D33 - sda
D32 - scl
gnd - gnd
3v3 - vcc
d5  - cs
D16- clk
d19 - miso
d23 - mosi

I connected also an LCD using a secondary SPI.

kami83, 2021-04-23:
@imaurus : I have some problems with a TTGO T7 Board. Can you post your sketch for the esp32 Dev? Thanks a lot.

Cu kami

ZePedro80, 2021-08-10:
Hi guys! How did you manage to get a proper 160x120 frame, if the library only works for 80x60 pixels (Lepton versions prior to 3.x)?

khaoulaMOS, 2022-02-01:
hello i dont know why the frame is like that 
[image]
 could you plz help me

ZePedro80, 2022-02-01:
> I would say that's a sync issue. I had a similar problem with my ESP32. I had to buy a Raspberry Pi and use it instead

NachtRaveVL, 2022-06-22:
Still not working, although I have seen some teams be able to modify the v1 codebase with STM and get it working, I've been busy with another project that has been taking precedence at the moment. I will return to this in time, for now tho, I would suggest looking at different options.

It would be nice to get this working on ESP32 though, as I am a huge fan of the MCUs.

JSpiner, 2022-06-28:
Hi, I'm trying with ESP32 board.
[image]
I connected lepton with [this pinmap](https://github.com/NachtRaveVL/Lepton-FLiR-Arduino/issues/8#issuecomment-757337449)
But it prints `LeptonFLir::checkErros lastI2CError: 2: Received NACK on transmit of address`
It also happens when disconnect all cables at lepton.

[image]
By any chance, I'm using lepton lev kit v2, is the pin-map different here?

NachtRaveVL, 2022-06-29:
Pinmap is a bit different on v2 breakout, as they have included the ISR pin for VSYNC.

Currently the library is undergoing a massive redesign and rework, but has been on the back burner while other things take precedence. I wanted to include a bunch of the various functionality that the Lepton 3.5 exposes but, it's just a lotttttt of unpaid work. Lots of writing slightly different methods over and over and over and over again, save nothing about the more fun parts.

I was also eyeing a new library, TurboSPI, but haven't dived into it enough just yet.

Really though, this project has been superseded by other stuff for quite some time now. It seems like no matter how many times I tell people there is just a constant instream of folk staring and posting, so.

Anyways, I may have time in August to look more into it. No promises. Current work takes precedence.

rmz-rider, 2022-12-06:
Hi,
I' m working on a project with thermal cameras. Saw, the last post is a few months old. So, is it still not possible to get 160x120 frame with Flir Lepton 3.5 using ESP32-S3 ?

thanks a lot

ramakrishna99, 2023-02-05:
> Hi, I' m working on a project with thermal cameras. Saw, the last post is a few months old. So, is it still not possible to get 160x120 frame with Flir Lepton 3.5 using ESP32-S3 ?
> 
> thanks a lot

Hi, I am working on similar project and getting similar issue "LeptonFLiR::checkErrors lastI2CError: 2: Received NACK on transmit of address". Please help if you resolved this issue.