NachtRaveVL/Lepton-FLiR-Arduino #9: Issue

strellis, 2021-03-12:
Thank you for the code. I keep getting the same issue below and I have no clue how to address it. Thank you for any help!

sketch\LeptonFLiR+Protected.cpp: In member function 'void LeptonFLiR::advanceNextFrame()':
sketch\LeptonFLiR+Protected.cpp:94:32: error: 'nothrow' is not a member of 'std'
         _nextFrame = new (std::nothrow) LeptonFLiR::FrameSettings(_lastFrame, _frameCounter++);

NachtRaveVL, 2021-03-12:
Oh that’s a good catch actually!

You should just be able to remove the (std::nothrow) part - that’s a C++
thing that makes it return null and not throw an exception if it cannot
allocate the buffer. It’s probably not there anymore because they probably
disable exceptions by default now.

Do note that the current version is incomplete. Wasn’t able to get it to
read frame buffers using the SPI library. Might be different for faster
chipsets but I wanted to rewrite it to take advantage of DMA with the STM32
chips. Just haven’t really gotten there yet in my life, with work and all.

Best of luck!

jaworskidrew, 2021-03-12:
I will be trying to make some progress on it this weekend; I'll make a pull request if I get anywhere, but I'm doubtful I'll get far. Right now, your code here is the closest to getting a Lepton 3.5 and Teensy 4.0 working together.