NachtRaveVL/Lepton-FLiR-Arduino #10: Lepton 3.5 frame rate and VSYNC

TinManAkshay, 2021-04-14:
Hi,

I am thinking to use these libraries, but had few questions in mind.

1. Have you implemented VSYNC functionality successfully in these libraries?
2. What is the max frame rate you could get?

Please let me know when you can.

Thanks,
Akshay

ArsenioDev, 2021-04-14:
Max frame rate is the ITAR-Compliant <9Hz (UNLESS YOU HAVE AN UNLOCKED LEPTON, which by the way are all but unobtanium lmao)

NachtRaveVL, 2021-04-14:
Honestly, back when I was heavily invested in writing this I was only able to go up to an Arduino Due, and even at the 84MHz it runs at barely was able to even get *a* frame, let alone anywhere near 9Hz. It would basically be unable to do the frame transfer, and quickly go from reading valid frame data to reading the EOF error code for desync (because it took too long to read the frame buffer). All the command functions worked though, so I figured that it just wasn't possible through the Arduino SPI library as implemented at the time (which I have heard is not the greatest).

I've seen people do custom SPI port manipulation stuff (very dependent on your board ofc) to get up to the transfer speeds needed to not desync, but that's kind of a cheap solution. I additionally was told by the makers of the FLiR chip that they would have to radically change their stack architecture to make it so you could read a frame without desync (and just allow a slower SPI transfer speed), so no luck there either. Ultimately I felt like the better solution would be to await a hardware DMA solution - such now exists ofc via STM32, but I just haven't really had the time to go back and work on this library as much as I would like to again.

As well, the fact that you can now do these kinds of FLIR applications on things like OpenMV kinda makes it a more-m00t point, anyways, especially if you want to offload the massive amount of processing speed needed to both read a frame and do any considerable frame specific image processing onto one microprocessor. I did want to still finish it off though, even if it meant using ESP32-CAMs, which I think has its own FLIR library as well (maybe a different language, can't remember, too lazy to Google). 

Anyways, if I ever do get back to finishing this, it would be basically be more like a tech demonstrator at that point since it would just be too risky for companies to depend on an external party (i.e. me) for any considerable amount of updates/bug fixes/etc. without a contract in place to begin with. Which I wouldn't want to do, anyways, since that's way too involved.

I'll come back to it eventually. Work kinda takes precedence for now.

Best of luck!

TinManAkshay, 2021-04-15:
So you are saying that these libraries are working good on ESP32 and Teensy so far? (OR) its still in progress??

Thanks
Akshay

NachtRaveVL, 2021-04-18:
Oh, it's still very much in progress. I was attempting to capture more of the up to v3.5 functionality of the chip, just because that at least was somewhat working through i2c. The SPI interface is probably enough now with the 300-400+ MHz chips we see on the Arduino market (if not the STM32 DMA stuff), but ultimately as mentioned the more specialized chipsets for the task of image processing (such as OpenMV and ESP32CAM) are much better options to be using than an Arduino.

As it stands, it's very much a pet project that I wanna finish it's just that I genuinely have very little time due to work taking precedence.

ArsenioDev, 2021-04-24:
Side note, I have a proof of concept that I forgot where I got from but runs at 9hz on the t3.2 to display. Gotta dig it out from my source directory sometimes

GoncaloRodrigues241, 2021-06-16:
Hi!!

I connected Lepton 3.5 to Adafruit ESP32 Feather but I can't read frames from IR camera. I followed @NachtRaveVL examples from his library but I still can't read the frames. I don't know if there is a frequency problems or not. I am using Serial.begin(115200) and Wire.setClock(400000).

Can you help me, please?

hvictor, 2021-10-01:
Hey guys, have a look at my NOCTIX-1 project. I have designed and manufactured a PCB based on a 200 MHz PIC32 do use the Lepton 3.5. With a 20 MHz SPI channel I read out the frame packets, and with a 100 MHz SPI channel I visualize the frames on a TFT at 11 FPS now. I am going to implement DMA support, so I expect that it will run faster... have a look at it